Wouter1/EMU-driver #131: Catalina install problem 0404 USB

wtnh, 2020-08-03:
Hi - I was excited to see this driver, so I dusted off my old 0404 and tried the install on Catalina on a Macbook Pro "late 2013".

Followed directions, except I could not see how to do the step which says "give full access to the driver installer". (I did give full disk access to terminal). This could be my issue, but not sure how to do that because the driver installer does not show up.

The 0404 shows up in USB devices but the driver is listed as the original EMU driver, which does not seem right.
`E-MU 0404 | USB:

  Product ID:	0x3f04
  Vendor ID:	0x041e  (Creative Labs)
  Version:	1.00
  Serial Number:	E-MU-0A-3F04-07D7081D-08926-STATION01
  Speed:	Up to 480 Mb/s
  Manufacturer:	E-MU Systems, Inc.
  Location ID:	0x14100000 / 10
  Current Available (mA):	500
  Current Required (mA):	2
  Extra Operating Current (mA):	0`

I re-ran the installer and had to answer "y" to a bunch of access questions this time (see attached script output).

[EMU driver install log.txt](https://github.com/Wouter1/EMU-driver/files/5018115/EMU.driver.install.log.txt)

Any hints appreciated! Thanks!

Wouter1, 2020-08-03:
@wtnh  if the control panel shows the EMU info like sample speed etc, it's properly installed.

The log file you attached shows that the EMU does not install  because /System/Library/Extensions is read-only.

This might be because you either did not do the csrutil or did not give full disk access to the terminal and driver installer. 

I think you can drag the installer program into the browser window after you click on the "+"

wtnh, 2020-08-03:
OK - I think it is working. I just did a listening test and the EMU functions fine for sound output. I am now testing the input using REW, an audio analysis program that I want to use for audio amp testing.

When I re-ran the installer, I answered "y" to each of the access override permissions.

What confused me is that in your instructions, it says to give disk access to the installer, but I had not downloaded it yet (that comes later in the instructions), so it was not visible in the list of apps to which I could give access.

So is it OK that in the list of USB devices that it shows up as the original EMU V1 driver, even though your driver is active?

Wouter1, 2020-08-04:
@wtnh  ok I fixed the install instructions to first download and unzip to fix that confusion.
Good to hear it works now!

>So is it OK that in the list of USB devices that it shows up as the original EMU V1 driver, even though your driver is active?

I can't change the USB Devices info, it's hardware

What matters is the EMUUSBAudio info in the About This Mac/System Report/Extensions. It should show the latest version (4.1.0 at this time)

wtnh, 2020-08-04:
Thanks for that - it does indeed show correctly in the Extensions report as 4.1.0. So everything is working despite my roundabout installation process (with the exception of the control panel needing to be started from terminal).

As an aside, I am experimenting with the REW application, with the goal of using the EMU as a distortion measuring tool. Attached is a screenshot showing the FFT performance of the EMU looped back on itself. Not too shabby for a 13-year-old interface I paid $150 for back then. So glad I was able to dust it off thanks to your great driver!

[image]

Wouter1, 2020-08-04:
Yes it's a known workaround for the control panel. I don't know any other way to fix it 

Almost 120dB -15dB = 105dB range, yes looks good. 

Very little at 50/60Hz, are you using something to suppress net hum etc?

wtnh, 2020-08-04:
Nothing special to suppress hum. I did try it with the Macbook on battery power only but saw no difference. I am on 60Hz power. The hump around 100Hz is interesting - I am wondering if the EMU wall wart is contributing some of that. It would be interesting to run the EMU on battery power to see if noise decreases further. Right now I am seeing THD of 0.00016% and noise of 0.0019%, so noise is dominant.

Attached is a better screenshot showing the FFT particulars.

[image]

wtnh, 2020-08-04:
Closing this issue - thanks for the help!

Wouter1, 2020-08-05:
@wtnh  thanks for the clarification. Yes I was thinking that you might be running the EMU on battery since I see no 50/60Hz residue. Are you using the original power adapter?  Let me know if you find any improvement with running on battery power!

wtnh, 2020-08-05:
Yup - that is with the original EMU power adapter. I am thinking of trying one of those cell phone 10,000 mA-hr 5V power packs, but I think they usually have some sort of buck/boost converter built-in, which would add hash to the output.

An ideal battery supply would consist of some LiPo cells and a linear 5 V regulator. I might throw something together and test that approach. Right now the EMU is "good enough" for testing (with REW) on a class-A amp I am building: https://www.diyaudio.com/forums/pass-labs/347697-hippie-trail-head-zombie-post6299801.html

That thread is a discussion about building clones of Nelson Pass's First Watt F7 amplifier.

Cheers

Wouter1, 2020-08-06:
Yes, a plain battery should be used, no switching PSU as then you get other types of noise.

Nice project. Good luck with that!

wtnh, 2020-09-14:
Just to follow up on this - I did try a battery pack which eliminated the weird 120 Hz noise caused by the EMU wall wart:
[https://www.amazon.com/gp/product/B082NKPPF3/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1](url)
It will run the EMU all day and does not seem to exhibit a lot of switching noise, at least in the audible band.

This is a nice little battery and it has a display showing state of charge.

[image]

Wouter1, 2020-09-14:
@wtnh  thanks for the update!
Can you post the noise spectrum with battery instead of PSU?

BTW I found that it even works on just 4 AA batteries. That may be even better as that will not contain any switching software (which that powerbank probably has)

wtnh, 2020-09-22:
Sorry for the slow response - here is an FFT spectrum with the EMU running on the above-mentioned battery pack instead of it's wall-wart. Note that the 120 Hz noise (and its harmonics) is gone. Also note that this shows the fundamental at -20db, so it is not exactly the same measurement setup. I have a common mode USB power filter with a choke and some caps, but have not had a chance to try that yet. Presumably, it would lower the noise floor a bit more.

[image]

Wouter1, 2020-09-22:
@wtnh  thanks!
That looks dramatically better than the adapter! 

That's also what I hear when using the power adapter versus batteries :-)

Looking forward to your test with your other measurements. 

Can you also measure when running straight on batteries (eg 4 AA cells), so that no switching supporession is needed ?

wtnh, 2020-09-22:
Before I do anymore testing, I want to stop using the phone plug inputs and start using the XLR inputs. The reason is that the phone inputs go through a pair of 5532 non-inverting buffers, while the XLR inputs bypass these completely. Therefore, the XLR inputs should exhibit lower noise and distortion (perhaps not a lot). I do not have any extra XLRs handy so I am ordering some.

Here is a schematic reverse-engineered by a modder a while ago showing the input arrangement. When a phone plug is inserted into the Neutrik jack, a relay switches the buffers into the input circuitry. With no phone plug inserted, the XLRs are connected.

[image]

Wouter1, 2020-09-23:
@wtnh 

Nice! So we then can compare and recommend users about XLR versus phone plug , and have detailed schematics too

Do you have a more detailed version of the schematic (the resolution is just too low to read the text)? Is this a SPICE model?

wtnh, 2020-09-23:
No - that is the only reference I could find to the schematic, unfortunately. You can make out most of the values if you zoom in on the picture. And you would have to manually re-create it in spice, I would think.

I found it here in the DIY audio forum: https://www.diyaudio.com/forums/diyaudio-com-wiki/162512-mu-0404-usb-mod-wiki-3.html#post5250447

That post mentions a forum DIYer named "sdiy" who I have not been able to track down. He created the schematic, apparently. Also, that thread discusses various mods people tried on the EMU, in particular, fixing the problems with the left channel caused by poor PCB layout.

While the 5532 was a good opamp in its day, bypassing it can only improve things and simply requires using the XLR pins.

Cheers

wtnh, 2020-09-23:
Found another thread in the DIY forum where sdiy discusses the schematics and the mods he was working on, FWIW:
https://www.diyaudio.com/forums/equipment-and-tools/222096-mu-0404-usb-2-0-artifacts-peaks-left-channel-7.html#post4175441

________________________________
From: Whit Turner <whit.turner@outlook.com>
Sent: Wednesday, September 23, 2020 9:54 AM
To: Wouter1/EMU-driver <reply@reply.github.com>
Subject: Re: [Wouter1/EMU-driver] Catalina install problem 0404 USB (#131)

No - that is the only reference I could find to the schematic, unfortunately. You can make out most of the values if you zoom in on the picture. And you would have to manually re-create it in spice, I would think.

I found it here in the DIY audio forum: https://www.diyaudio.com/forums/diyaudio-com-wiki/162512-mu-0404-usb-mod-wiki-3.html#post5250447

That post mentions a forum DIYer named "sdiy" who I have not been able to track down. He created the schematic, apparently. Also, that thread discusses various mods people tried on the EMU, in particular, fixing the problems with the left channel caused by poor PCB layout.

While the 5532 was a good opamp in its day, bypassing it can only improve things and simply requires using the XLR pins.

Cheers

Wouter1, 2020-09-24:
@wtnh thanks for the additional checking. 
Yes you can guess most values. But I wasn't sure about the opamp types

I can't see the pics on that forum, apparently you need an account for that. 
Was there any higher resolution version there?

wtnh, 2020-09-24:
The best I can do is this one. If you have some image viewer app, you should be able to zoom and read the values.

The Two input opamps are 5532 and the rest are NJM2068 duals. Datasheet is here: https://www.mouser.com/datasheet/2/294/NJM2068_E-364244.pdf

I have not poked around for LTspice models for the NJM2068, but I am sure they exist and I know they do for 5532.

Getting an account on DIY Audio is pretty painless - and there is a treasure trove of audio-related info - and some very smart people too (and some clueless as well 🙂). There is a whole forum dedicated to Nelson Pass's stuff and he contributes regularly. Also guys like Bob Cordell. I recommend it.

[cid:ea35a456-bd3d-454d-a679-2ddc0237398c]

wtnh, 2020-09-24:
Oops - email did not push the image to Github. Here is the best image I can get from the forum.

[image]

wtnh, 2020-09-24:
BTW - if you are interested in the topology of the opamp stages, you can read more about it in Bill Whitlock's paper. He invented the differential amplifier bootstrapping technique designed into the EMU. It is patented, so I am guessing Creative had to pay something to use the circuit. But it is very clever and greatly increases CMMR and input impedance.
http://www.jhbrandt.net/wp-content/uploads/2014/11/Design_of_High-Performance_Balanced_Audio_Interfaces.pdf

Wouter1, 2020-09-24:
@wtnh yes that looks interesting, they are putting an offset which is the average of the L&R signals into both opamps' feedback. Strange, thanks for figuring that out and finding that paper! I'll check it out

wtnh, 2020-09-24:
Actually, that is not L+R; it is only for one channel. In other words, the schematic is showing just one channel. The opamps are arranged in differential pairs. The bootstrap opamps (the ones pointing in the opposite direction) take the summed signal from the differential pairs (which in a perfect world would be zero), invert the error signal and feed it back to the inputs of the pair, effectively canceling out any imbalance between the pair. This raises the CMRR and input impedance of the pair. It is extremely clever (which is why it received a patent).

I think it speaks well to the care that went into designing the EMU 0404 - they could have cheaped out like so many other interfaces have. To this day, it stands up well against more "modern" interfaces. It's just too bad Creative dropped support for the drivers long ago.

If you read through the threads on mods, there are some things that will definitely improve performance, especially the internal power supplies. I have not tackled any of those, since it suits my present purposes (distortion measurement of things I am building).

I have built my own Raspberry Pi - based D to A converter/streamer and I use that for music listening, and it does better than most USB-based devices since it does not depend on clock reconstruction of the USB stream, but that's a whole other story 🙂.

Wouter1, 2020-09-24:
Ah yes of course, thanks for pointing out. I was mistakenly interpreting the L and R of XLR as being L and R channel. THe name XLR seems just wrong now :-) 
I read some of that paper. If I get it right the best way is to have very high input impedance. What I don't understand yet is why they then not just connect it straight to an opamp input and get rid of all resistors they have. 

Regarding modding this, might be useful but I have the impression the noise I hear is mainly from the mics, not from the amps. But I could be wrong there

Why don't you use your EMU for listening? Is the DA part not so good? The EMU has its internal clock anyway, it does not 'reconstruct' from the USB stream.

wtnh, 2020-09-24:
The EMU is a "kitchen sink" interface and does most things extremely well, especially given the fact that it came out 14 years ago. It does A to D, D to A, has mic inputs, SPDIF, USB, etc. But D to A technology has moved on. The DAC clock jitter in the EMU was spec'd out as I recall, as being < 1 ns. Modern high-end DAC clocks are more like 1 ps jitter or less. That's 1000 X better.

The D/A converter I am using interfaces to a Raspberry Pi through the I2S bus and provides the clocking back to the Pi. So, the stream is under the direct control of the DAC clock, not the other way around. This is referred to as "master" clock mode and is a more efficient solution (IMHO anyway) than adding the complexity of a FIFO or the instability of a PLL clock scheme. The DAC still has to have good clocks, of course, because jitter translates into noise and distortion.

Another thing that helps is I am using an isolator between the Pi and the DAC, so the DAC is galvanically isolated from the Pi and has its own separate power supply, which is LiPo batteries - so it is very quiet.

The EMU can sound pretty good playing digital files, especially at higher bitrates, but a purpose-built D/A converter can do much better these days.

Cheers

wtnh, 2020-09-25:
Here are two loopback test runs comparing single-ended (output and input) to differential (balanced output with XLR input) setups. Since balanced outputs are 2x the voltage of single-ended, I adjusted the single-ended run to match the level with the differential test. There is not a huge difference, but the differential setup exhibits lower THD. There is hardly any difference in noise so most of that must come after the 5532 buffers. In fact, noise dominates the measurements, and to get rid of it would require some mods to the EMU. I am also now using a small USB power filter between the battery supply and the EMU power input, just to make sure not much hash is getting into the EMU.

SINGLE ENDED:
[image]

DIFFERENTIAL:
[image]

Wouter1, 2020-09-28:
Surprisingly, the one on the battery pack and line input seems a tiny bit cleaner even?
It goes down to below -140 while these ones go to -130 so overall 10dB less noise with the line inputs?

wtnh, 2020-09-28:
Sorry - I should have labeled the most recent FFTs better. BOTH of them are using the battery supply and hash filter.

The hash filter is a bit of an unknown quantity - came from eBay. It has an input 220 uF cap followed by a small common-mode toroidal ferrite choke (about 1/2-inch diameter with I would guess 20 turns on each winding), followed by a 1000 uF cap on the output. There are USB jacks on either end. I am guessing that is doing a decent job of noise reduction from the battery. I might throw a scope on it to see if there is a difference.

The difference in noise between single-ended and balanced setups is almost negligible - looking at 3 parts per million difference, which is tiny and could vary depending on the exact levels and battery condition. I did try to run the tests far away from any electrical sources (lights, power outlets, etc.) and the two runs were within 5 minutes of each other. The only thing I did was to swap cables.

The single-ended test was using a well-shielded mono guitar cable, but the plugs are not that great (not gold) and pretty old. They could benefit from some de-ox, I am sure. Same with the jacks on the EMU. If I wiggle the connections, the readings go crazy then settle. So, cables, shielding, contact noise all play into measurements at these tiny levels.

The most interesting part of the experiment was bypassing the 5532's, resulting in significantly lower THD. THD is now so low that it makes the EMU a pretty good measuring instrument, at least for 1KHz sine waves 🙂.

The noise is still a problem and swamps the THD.

A lot of people in the DIY forum are using Focusrite Solos for this sort of REW distortion measuring, but they show a higher THD and noise than I am getting with the EMU.

Another thing I might try at some point is building an ultra-low noise/distortion sine wave generator - I have a design done by JIm Williams (of Linear fame) some time ago which has the best numbers I have ever seen. Some folks are using a sine generator from Akita for the same purpose. That would eliminate the output of the EMU from the equation as well as the internal D/A converter (which could be contributing to the noise floor).

Cheers

Wouter1, 2020-09-28:
@wtnh Ah what I also missed is that you changed the scale. In the earlier pic we have only 50-100Hz at the far left, while in the new pics you allocate more space for the 20-100Hz

It might make sense to plot only the range from -80..-160dB as that is the range of interest. I suppose the single peak is normalized at 0dB anyway, no need to show that.

The hash filter is confusing. Is this a filter for the power supply? Or for the USB cable? I assume for the PS but why would you add USB connectors to that

The higher harmonics, I suppose they come from the AD and DA converters? 

FAIK, with 24 bit converters the noise in a perfect converter is around 2^-23 = 10^-7 which is equivalent to 140dB (assuming your max signal is at 0dB). So these plots may be already pretty close to the limit. 
Did you have the signal at 0dB ? 

Also, if the signal at 0dB would be 1V, then the noise in that case would be at 0.1uV. FAIK  0.1uV is pretty tough to handle.

Wouter1, 2020-10-06:
@wtnh I found that batteries also are a noise source. So although better than a switching power supply, they are not perfect.

It Might be an idea to try a low noise power supply to see if noise drops down even further

wtnh, 2020-10-06:
Yes - I did not bother to fool around with the scales on those two runs. I should be more deliberate next time - but since the conditions were the same for both single-ended and differential, at least there is an apples to apples comparison.

Also, note that the fundamental is -14.85 down (from zero). I did this because it results in the lowest distortion - if I boost it up toward zero, distortion increases, and I am not sure what part of the EMU is causing this (could be the D to A, the output drivers, etc. So, to get a true indication of the harmonic levels, you would need to add 14.85 db to the levels in the charts in order to normalize.

I'll post a picture of the USB filter. I am using it in line with the battery supply and the EMU.

I agree it would be interesting to try a low noise supply, but if it uses a transformer connected to the AC mains, there are additional issues with noise coupling.

A better choice might be to follow the battery supply with a supercapacitor.

But at that point, the internal boost supply in the EMU will swamp any improvements.

wtnh, 2020-10-06:
Here is the power setup, along with the USB power filter:
[image]
[image]
I got the filter on eBay a couple of years ago, originally to put between a wall wart and a Pi Zero music streamer. As I mentioned it is simply a choke with two electrolytic capacitors.

Wouter1, 2020-10-06:
@wtnh  I copied your diagram of the input amps to the docs directory  as it seems important documentation to me.

Wouter1, 2020-10-06:
@wtnh 
> Also, note that the fundamental is -14.85 down (from zero). I did this because it results in the lowest distortion - if I boost it up toward zero, distortion increases, and I am not sure what part [...]

Ah another important point for realizing the best possible noise performance.
I noticed that at some point roughly at 70% of full setting the input level seem suddenly to get more sensitive.
Could this be related to this? 
Could there be some second amplifier kicking in at that point?

Is the choke having a significant effect? 

I am thinking of making a single plot overlaying all these results so that the differences are clear in one look. 

Would it be possible to post the raw data so that I can throw all plot points through Mathematica for such a plot?

Also do you have plots when the EMU is running on plain batteries? The non-switching type I mean. In the past I used a 4-pack NiCd cells and they were great, NiCd has lowest noise, but unfortunately you can't get them anymore... The NiMH have more noise. 

I do have similar measurements myself as well, I just dug them up from my archives (2015). These plots are based on an tables like this 
```
Frequency (Hz)	Level (dB)
2.691650	-92.529640
5.383301	-123.497314
8.074951	-124.782196
10.766602	-125.456329
13.458252	-125.506355
16.149902	-126.773232
18.841553	-128.253525
21.533203	-128.495392
24.224854	-128.741226
26.916504	-128.646927
29.608154	-129.220505
32.299805	-129.836273
34.991455	-130.184021
```

I think these are exported from Audacity.

But if you could provide anything similar I can make some nice plots.

I'd have to think about the normalization,  I suppose all values should  be relative to 0dB?

Wouter1, 2020-10-06:
@wtnh  I did a lot of measurements back then. I checked the effect of
* batteries, empty versus full, and power adapter
* with or without mics, with XLR plug but no mics attached to the plug,
* headphone attached or not
* different gain settings
* sample rate
* Computer attached to mains vs on battery
* shielding the whole setup in metal or not
* inside vs outside

I think I should make a wiki page on how to get the lowest noise

wtnh, 2020-10-06:
The input level pot is actually a gain control, not an attenuator. I was able to get the best results by keeping the control at the lowest level - I am guessing that results in unity gain in the input amplifier. The inputs on the EMU are VERY sensitive. I'll have to experiment with REW to see how to export data - I think it has that capability, but I have not used it.

Here is a diagram I posted in the DIY Audio REW forum showing the testing setup I have used for testing with the EMU:

[image]

Wouter1, 2020-10-06:
@wtnh  For the tests here we are basically testing the EMU itself.

So I suppose this picture is not what you use then?

The tests that I did were either using mics straight connected to the XLRs
or line out loopback directly into line in jack plugs

The main problem then is that I'm measuring the output and input performance of the EMU together (assuming the wire and connectors do not add to the distortion).

I would need a separate, better device to test them independently 

Also the voltmeter might also inject some noise, because it has switching devices that might reflect back on their probe lines. Even just the extra wires could cause extra noise.

Wouter1, 2020-10-06:
@wtnh 

> I am guessing that results in unity gain in the input amplifier. 

I would not assume that. I see all kind of weird ratios in your schematic. For instance in the first XLR stage I see 3K versus 20K (max setting of logarithmic) +1.8 ohm. So my first impression this would be a 1.15 ... 1666x amplifier

wtnh, 2020-10-06:
When I did the loopback tests, I just connected the left channel output to the right (b) inputs. The single ended test was with a mono cable (tip-ring) and the differential test was with a stereo cable to the XLR input.

The voltmeter will have very little effect with low-Z loads - typically 8 ohms for power amps. This is because the input impedance of the meter is in the mega-ohm range. Also any common mode noise injected by the meter will be greatly reduced by the CMRR of the balanced input.

wtnh, 2020-10-06:
Yes - I don't know for sure what the gain the circuit has - also I am not sure I 100% trust that schematic or the values.

It might be interesting to find an old EMU 0404 (maybe not working) on eBay and rip it apart to validate the schematic and values. Some of the parts may have to be "lifted" to get accurate readings.

What I do know is that if I don't attenuate the input enough or crank the gain pot up, distortion goes through the roof, so it is very easy to overload the EMU inputs. That is why I have a pretty hefty attenuator in my test setup.

wtnh, 2020-10-06:
Oh - and the voltmeter can be removed once the levels are set, prior to doing any critical THD tests.
________________________________
From: Whit Turner <whit.turner@outlook.com>
Sent: Tuesday, October 6, 2020 2:28 PM
To: Wouter1/EMU-driver <reply@reply.github.com>
Subject: Re: [Wouter1/EMU-driver] Testing noise levels (#131)

When I did the loopback tests, I just connected the left channel output to the right (b) inputs. The single ended test was with a mono cable (tip-ring) and the differential test was with a stereo cable to the XLR input.

The voltmeter will have very little effect with low-Z loads - typically 8 ohms for power amps. This is because the input impedance of the meter is in the mega-ohm range. Also any common mode noise injected by the meter will be greatly reduced by the CMRR of the balanced input.

Wouter1, 2020-10-06:
@wtnh 

But if you are measuring in the -140dB range, then this would be 1uV against 1V. Or  8ohm versus 8Meg which might be comparable to your meter. Also I'm not sure if this would meter resistance would be relevant if the wires themselves to the meter are acting as pickup antennas for RF noise. If that RF noise would cause a few uA of current noise it would be enough to raise your noise floor. But I'm not an electric engineer, I might misunderstand something 

But if you remove it after initial adjustment anyway it's not an issue.

Why lift the parts, are the values not just stamped on the parts?

wtnh, 2020-10-06:
The pictures I have seen of the interior lead me to believe it might be difficult to read the markings. It’s all surface mount parts and they are tiny. I have looked around on the web to find official schematics but there seem to be none.