Wouter1/EMU-driver #132: Big Sur compatibility problem

oxygen311, 2020-11-13:
The driver can't be installed on Big Sur as is because of new [encryption features](https://www.reddit.com/r/MacOSBeta/comments/he4umc/how_to_make_root_volume_writeable_in_big_sur/). 
Just mounting root for writing isn't easy, but even it's mounted, I'm not sure the driver would work.

Wouter1, 2020-11-13:
I have no Big Sur here so I can't confirm or help here. 

The link you gave is on how to make root writable. Is the csrutil not working?

Please post here if yo figure out how to get it working, other users will have the same issue if they want to move to Big Sur.

oxygen311, 2020-11-13:
> @oxygen311 I have no Big Sur here so I can't confirm or help here.

Do you plan to upgrade to Big Sur?

> The link you gave is on how to make root writable. Is the csrutil not working?
> Please post here if yo figure out how to get it working, other users will have the same issue if they want to move to Big Sur.

csrutil disabling is working, but it doesn't help to mount root. Other ways seem too confusing for me right now, maybe I will have time and try them later this week.

I updated to Big Sur with the driver installed in Catalina, but the driver stops working. So I suggest not to update for users who want this driver to work.

Wouter1, 2020-11-13:
Ok so the problem is not a encryption feature but a write permission issue ? You can't copy the driver into the extensions folder?

No upgrade plans here. For me, upgrades often cause more issues than they solve. So I can't help you with the write permission issue.

oxygen311, 2020-11-13:
I've mounted volume to write in it. If even it's done, you suppose to make changes to `/Volumes/Macintosh SSD 1`, but not root folder, so can't simply install with the available script.
I've tried to run the script manually changing `/` to the mounted volume. And got an error while loading kext:
> `Kernel request failed: (libkern/kext) not loadable (reason unspecified) (-603946989)`

Wouter1, 2020-11-13:
@oxygen311  Just to check and get some details. What message do you get if you do this
```
sudo kextload /System/Library/Extensions/EMUUSBAudio.kext  
```

oxygen311, 2020-11-13:
I was getting this error:
> `Kernel request failed: (libkern/kext) not loadable (reason unspecified) (-603946989)`

Maybe it's because I was trying to load kext from the mounted drive. 
I'm afraid of booting from this mounted drive as the main one, is there any change that mac won't run with inappropriate kext? (I'm not really into such things)

Wouter1, 2020-11-13:
@oxygen311 
FAIK the extension has to be in /System/Library/Extensions, so it must be on the boot disk. So mounted drive won't work unless that's the drive you booted from.

>is there any change that mac won't run with inappropriate kext

The ikext loads only when needed, so when the EMU is connected and turned on. If you don't have the EMU connected or it's off, then the kext will not even be loaded. 

If the mac won't boot, you can turn it off, unplug the drive and reboot using the built-in drive. Or you can keep the option key pressed while booting up, then you can select the boot volume.

oxygen311, 2020-11-13:
[image]

Did all tricks to load files into the system folder. But still getting this error.

vasukovich, 2020-11-14:
> The driver can't be installed on Big Sur as is because of new [encryption features](https://www.reddit.com/r/MacOSBeta/comments/he4umc/how_to_make_root_volume_writeable_in_big_sur/).
> Just mounting root for writing isn't easy, but even it's mounted, I'm not sure the driver would work.

Hi. After update to Big Sur 11.0.1 I used this app https://github.com/chris1111/Kext-Droplet-Big-Sur to install the kext. And this worked well for me. Now I have EMU 0204 working in Big Sur. Maybe this will help to you also.

oxygen311, 2020-11-14:
> Hi. After update to Big Sur 11.0.1 I used this app https://github.com/chris1111/Kext-Droplet-Big-Sur to install the kext. And this worked well for me. Now I have EMU 0204 working in Big Sur. Maybe this will help to you also.

Thank you for your advice!
I installed `EMUUSBAudio.kext` from `v11` folder and restarted my mac as tool asked.
But E-MU 0404 is still don't detecting on my MacBook :(

Did you do something else besides that?

vasukovich, 2020-11-14:
> > Hi. After update to Big Sur 11.0.1 I used this app https://github.com/chris1111/Kext-Droplet-Big-Sur to install the kext. And this worked well for me. Now I have EMU 0204 working in Big Sur. Maybe this will help to you also.
> 
> Thank you for your advice!
> I installed `EMUUSBAudio.kext` from `v11` folder and restarted my mac as tool asked.
> But E-MU 0404 is still don't detecting on my MacBook :(
> 
> Did you do something else besides that?
> Did you install MIDI Driver?

So I did all this steps:

1. reboot in recovery mode (hold cmd-R while booting)
2. go to utilities-terminal
3. enter this command: csrutil disable
4. reboot your machine
5. under "System Preferences > Security and Privacy > Privacy" grant full disk access to terminal and the driver installer
6. In a terminal do sudo mount -uw /

the last step failed.

Then I downloaded the app from previous post to install the kext.

After the app installed the kext you need to go to system preferences-security and privacy and in general tab press the button to confirm usage of newly installed kext. Like you do when install apps not from authorised sources.

oxygen311, 2020-11-14:
> After the app installed the kext you need to go to system preferences-security and privacy and in general tab press the button to confirm usage of newly installed kext. Like you do when install apps not from authorised sources. Without this last step it doesn't works.

It's weird, but I don't have this button in preferences, I tried multiple times and don't understand how to summon this button.
Did it ask you after installing kext or after reboot or after connecting sound card?

Wouter1, 2020-11-14:
@vasukovich  thanks for chiming in and giving suggestions. 

If I understand you all right, all the instructions for Catalina work except the last one, the mounting of / for writing.

Apparently Apple changed the OS to a encrypted, read-only system. 

https://developer.apple.com/forums/thread/649832
https://twitter.com/EBADTWEET/status/1275454103900971012

If I get it right, you need to make a new boot system called "snapshot" that includes the new kext. 

I did not read the details but it seems you can't re-enable security on your customized 'snapshot'. 
I can't test anything of all this.

Wouter1, 2020-11-14:
It would be great if someone could report on how to install the kext without need of a 3rd party tool like Kext-Droplet-Big-Sur.

oxygen311, 2020-11-14:
> After the app installed the kext you need to go to system preferences-security and privacy and in general tab press the button to confirm usage of newly installed kext. Like you do when install apps not from authorised sources. Without this last step it doesn't works.

I tried again and now it worked and card is working! Thank you so much!

Maybe that's because I changed the version to `v9` while installation or because of card was ON and connected while installation. Anyway, after that, I installed `v11` and it's asked that thing at the end of installation and now everithing is working.

> I did not read the details but it seems you can't re-enable security on your customized 'snapshot'.
> I can't test anything of all this.
I tried this way, didn't work for me. You can see on my screenshot that kext was in the System folder, but it didn't help it to work.

vasukovich, 2020-11-15:
> > After the app installed the kext you need to go to system preferences-security and privacy and in general tab press the button to confirm usage of newly installed kext. Like you do when install apps not from authorised sources. Without this last step it doesn't works.
> 
> I tried again and now it worked and card is working! Thank you so much!
> 
> Maybe that's because I changed the version to `v9` while installation or because of card was ON and connected while installation. Anyway, after that, I installed `v11` and it's asked that thing at the end of installation and now everithing is working.
> 
> > I did not read the details but it seems you can't re-enable security on your customized 'snapshot'.
> > I can't test anything of all this.
> 
> I tried this way, didn't work for me. You can see on my screenshot in reply above that kext was in the System folder, but it didn't help it to work.
> 
> Anyway, great that it can be run in Big Sur at all.

Great news!

highcat, 2020-12-19:
Thank you so much guys, I finally was able to make 0404 work on Big Sur 11.1
1. Enabled write access like described here https://egpu.io/forums/mac-setup/macos-up-to-11/
2. Fixed MacOS user password like described here https://appletoolbox.com/cant-enter-your-password-in-macos-big-sur-heres-how-to-fix-it/#Why_canrsquot_I_enter_my_password_in_macOS_Big_Sur 
3. Installed the extension using Kext-Droplet-Big-Sur from v11
4. Allowed permissions and restarted.

The first time I couldn't see the card somehow, but then repeated steps (3) and (4), opened E-MU control panel, and now it's working!

Wouter1, 2020-12-20:
@highcat  thanks for the detailed instructions.
I still hope though that someone can figure out how to install without 3rd party tools

I won't have access to big sure any time soon, it will probably after the corona circus only that I will try it. I can't risk any damage to my machine right now.

jefazo2k5, 2021-01-13:
Works in Bigsur!!!!! 

With disabled SIP.

copy kext on System/Library(use your volume and folder) using terminal:

`-cp -R /Users/User1/EMU-driver-master/EMUUSBAudio.kext /Library/Extensions`

`chown -v -R root:wheel /Library/Extensions

`-cache touch /Library/Extensions`

`-kextcache -i /`

-Appears:

_Rebuilding local auxiliary collection
kmutil done
Executing: /usr/bin/kmutil install --volume-root / --check-rebuild_

-Go to preferences Panel/System & Privacity  and Grant permision for EMU driver.

-reboot.

Working , THANKS wouter1!!

Wouter1, 2021-01-13:
@jefazo2k5  thanks for figuring out how this should be done without any external 3rd party tools!!

I will look into incorporating your code into the build script so that others can install it with minimal hassle.

roddy20, 2021-01-13:
about installer
`sudo cp -r EMUUSBAudio.kext /System/Library/Extensions`
will not work at Big Sur, because /System is protected
I did manually `sudo cp -r EMUUSBAudio.kext /Library/Extensions`
System asked to confirm unsigned extension from unknown developer, YES, password, and reboot
thats all

roddy20, 2021-01-13:
> `cp -R /Users/User1/EMU-driver-master/EMUUSBAudio.kext /Library/Extensions`
> `chown -v -R root:wheel /Library/Extensions`

its Ok, but no need for chown, "sudo cp" works with correct rights and owners

Wouter1, 2021-01-13:
@roddy20 @jefazo2k5 

thanks for the additional input.

Ok, so the only major change is that I now need to sudo copy to ```/Library/Extensions``` instead of ```/System/Library/Extensions``` ?

What is this ```cache touch /Library/Extensions``` for?

roddy20, 2021-01-13:
touch /Library/Extensions will update the "modified at "  time of /L*/E* and force the system to re-create caches
it useful but not strongly required

roddy20, 2021-01-13:
> Ok, so the only major change is that I now need to sudo copy to `/Library/Extensions` instead of `/System/Library/Extensions` ?

yes, and this works with older systems too

Wouter1, 2021-01-13:
@roddy20 yes I remember it was recommended earlier but it was not required so I did not change the installer.
This seems the first OS version that prohibits using /System/Library

Wouter1, 2021-01-22:
I borrowed a mac from my work with Big Sur

macbook air
2.2GHz dual core i7
8GB 1600MHz DDR3
Graphics Intel HD Graphics 6000 1538MB
Big Sur 11.1

I installed the driver like this
 * disable csrutil as usual
 * cd v11
 *  sudo cp -R EMUUSBAudio.kext /Library/Extensions
 
After a short wait a popup appears that a new kext is there.
Click to go to security & privacy preferences
In security & privacy panel you see at bottom "System software from developer 'Unidentified - EMUUSBAudio' has been updated". Click on allyw
Then reboot the system

After that the EMU jost works fine. I can select it in the audio preference panel and play back music with quicktime.

I played succesfully
* Some 8 minutes of pop music, at 96kHz
* Some 10 minutes of classical music, at 192kHz

Both sounded fine, I did not notice any distortions in the sound

Wouter1, 2021-01-22:
The installer does not work for Big Sur because apparently Apple changed the behaviour of {{{sw_vers}}}

{{{sw_vers -productVersion}}} used to return something like {{{10,11,5}}}

It now returns {{{11.1}}}

Wouter1, 2021-01-22:
@oxygen311  @vasukovich  @jefazo2k5  @roddy20  I fixed the installer and the insructions for install.

Here it now installs fine on big sur as well using the standard installer.

Please let me know if there are still issues with the installer. I assume this is fixed but if necessary we can reopen this.

closed 2021-01-22 by Wouter1