Wouter1/EMU-driver #133: playback stops when I open Safari

roddy20, 2021-01-15:
Hello!
a strange problem
playing music with iTunes, its Ok if I don't touch anything
when opening Safari, playback stops and EMU is not available until I replug it
logs attached

[EMU error.zip](https://github.com/Wouter1/EMU-driver/files/5818896/EMU.error.zip)
[EMU Error2.zip](https://github.com/Wouter1/EMU-driver/files/5818897/EMU.Error2.zip)

Wouter1, 2021-01-15:
Hard to tell from the logs.... I'm not sure what is going on on your machine. I see that something is resetting the driver in the 2nd log. 
I read that safari is playing silence. Maybe this is related to the problem?

roddy20, 2021-01-15:
How can I do more detailed test?
it seems to me, driver was stopped when CPU load exceed some limit and never run again until replug

Wouter1, 2021-01-15:
If safari somehow completely locks up the CPU log enough, then the EMU device might run out of its buffer and I think audio will stop then.
Maybe I can see more in your logs but I have no time right now.

roddy20, 2021-01-15:
There is no problem of that kind with old official driver (for Lion?)
but it can playback only, no recording, in modern macOS (( recording is very distorted
with your version recording works Ok

Ok, let me know if more logs or any reports will be needed

Wouter1, 2021-01-15:
@roddy20 Can you give the exact OS, what  machine (type of mac, year) and how you connect (direct to USB2/3/... or via hub) etc

roddy20, 2021-01-15:
> @roddy20 Can you give the exact OS, what machine (type of mac, year) and how you connect (direct to USB2/3/... or via hub) etc

MacPro5,1, 2010, direct to USB, I use Sierra 10.12.6 (and I have Big Sur 11.1 it is installed to another drive in order of experiment with key -no_compat_check just for testing)

Wouter1, 2021-01-15:
@roddy20 so problems occur on Sierra?
This mac has native USB2 ports so there should not be a compatibility issue.
Strange, people have been using my driver for years a.o, on sierra. Cataline, the currently common version, is already 4 versions further, and being phased out.

So that makes me wonder, have you been using sierra for 4 years now with the EMU and suddenly this driver issue appeared?

Is it possible that Safari somehow got updated to something not excactly compatible with Sierra?

roddy20, 2021-01-15:
> So that makes me wonder, have you been using sierra for 4 years now with the EMU and suddenly this driver issue appeared?

1) EMU is not the only one device, and I don't use it and your driver full-time 7/24
2) problem is not permanent,  it occurs sometimes, once a hour, maybe earlier, maybe later
"might run out of its buffer" I think it is very close to truth
3) it never happen when Mac is just playing music, always related with higher CPU load

Wouter1, 2021-01-15:
@roddy20 
@2 the driver buffer size is only enough for a split-second amount of sound. So it does not suggest me what the problem can be. I'm not sure about run times of hours, I usually test up to 10 minutes and do recordings of maybe 20 minutes. But again, there are many users and I never heard this problem
@3. What happens if you start safari right away after boot (so no other sounds produced before that time)? And what if you first start safari and then turn on the EMU? 

What version of safari do you have?

roddy20, 2021-01-15:
> @3. What happens if you start safari right away after boot (so no other sounds produced before that time)? And what if you first start safari and then turn on the EMU?

it will work OK
playback stops when I do something else during playback, I mean something with high CPU requirements
Safari is 12.1.2 (12607.3.10) default for Sierra

roddy20, 2021-01-15:
>I'm not sure about run times of hours, I usually test up to 10 minutes 

I ll try to find a way to reproduce the problem immediately
maybe some kind of Geekbench CPU test with 100% CPU load (?)

Wouter1, 2021-01-15:
Ok, it seems you are 100% sure about the 100% CPU.

Yes it's possible in theory, as I suggested, if some process with  high priority would eat up all CPU then the EMU would run out of buffer. As I said the driver buffer is tiny, in a matter of a split second the buffer would be empty and playback would stall. 

The key is 'high priority'. There should be no high prio process hogging the complete CPU for so long (a split second is very long in terms of CPU cycles). The interrupt handling has priority over normal user processes, but maybe safari is not programmed cleanly, or maybe some other driver does have higher prio and is not being fair.

So, could this be caused by another system extension? Some extension you installed (as part of some software)?

roddy20, 2021-01-15:
Problem appears with official old driver too
But it is very rare, and it can restore playback in a second after the error without unplug

About 3rd party kexts and software , I am not sure ... I ll reinstall clean macOS in some days, only macOS + your driver and macOS + official driver and I ll compare the behavior and let you know

Wouter1, 2021-01-15:
Maybe you can install osx on a scratch disk, just for testing? I think it can even be a USB stick. Just a fresh OS and then install just the EMU driver.

Smuteee, 2021-01-16:
Hi, just my two cents to your interesting discussion. First a very big thank you wouter for your ongoing support and driver, i would happily send you some money for making it possible to still use my beloved emu 404 usb with the various iterations of mac operating systems through the years. I use it for audio playback in apple music/vlc as well as playing vitual instruments (vst/au)in logic/cubase/unify in a quasi life situation with attached midi keyboards. For this application it is essential to have a very small buffer to reduce latency, so that the sound produced has no lag from pressing the keys. if you use cpu demanding vst it can cause the emu or driver to stall (audible crackles ) or stop working (no audio) with buffer settings to low. This behaviour i remember experiencing also with old original drivers from emu back in the day. All of this is well known and explains the problems adressed in this threat. Just for listening and recording/measurement You can set a much bigger and in that regard much more „real time“ safe audio  buffer without any performance issues, for example 50ms lag when starting playback is barely noticeable. You can adjust this setting with any serious audio application on the mac. Even some free apps can do it. Sorry for the long post. Thank you, Andi

roddy20, 2021-01-16:
> Maybe you can install osx on a scratch disk, just for testing? I think it can even be a USB stick. Just a fresh OS and then install just the EMU driver.

Ok, I do this next days
found a simple CPU stress test for 100% CPU load
`yes > /dev/null & yes > /dev/null & yes > /dev/null & yes > /dev/null &
killall yes
`

roddy20, 2021-01-16:
one more strange message in log
16:06:53.781365 +0200	kernel	** writeCompleted bad result e00002ee
16:06:53.845404 +0200	kernel	001458.004582 EHC2@fa000000: AppleUSBEHCI::removePeriodicListEntry: unable to find isochronous list element <private> for frame 779
16:06:53.845469 +0200	kernel	** writeCompleted bad result e00002ee
time is approx the same when playback stops and EHC2 is the port with EMU

roddy20, 2021-01-16:
I cannot reproduce the problem when I want it 
it happens randomly ((

roddy20, 2021-01-16:
one more time
2021-01-16 16:37:00.301257+0200 0x14b      Default     0x0                  0      kernel: (EMUUSBAudio) ** writeCompleted bad result e00002ee
2021-01-16 16:37:00.365171+0200 0xce       Default     0x0                  0      kernel: (AppleUSBEHCI) 001126.932693 EHC2@fa000000: AppleUSBEHCI::removePeriodicListEntry: unable to find isochronous list element <private> for frame 142
2021-01-16 16:37:00.365244+0200 0x14b      Default     0x0                  0      kernel: (EMUUSBAudio) ** writeCompleted bad result e00002ee

roddy20, 2021-01-16:
and here if Official EMU driver, the same error
2021-01-16 18:19:34.047136+0200 0xc9       Default     0x0                  0      kernel: (AppleUSBEHCI) 005499.308794 EHC2@fa000000: AppleUSBEHCI::removePeriodicListEntry: unable to find isochronous list element <private> for frame 499
2021-01-16 18:19:34.055145+0200 0xc9       Default     0x0                  0      kernel: (AppleUSBEHCI) 005499.316802 EHC2@fa000000: AppleUSBEHCI::removePeriodicListEntry: unable to find isochronous list element <private> for frame 507
2021-01-16 18:19:34.063140+0200 0xc9       Default     0x0                  0      kernel: (AppleUSBEHCI) 005499.324797 EHC2@fa000000: AppleUSBEHCI::removePeriodicListEntry: unable to find isochronous list element <private> for frame 515
2021-01-16 18:19:34.071137+0200 0xc9       Default     0x0                  0      kernel: (AppleUSBEHCI) 005499.332797 EHC2@fa000000: AppleUSBEHCI::removePeriodicListEntry: unable to find isochronous list element <private> for frame 523
2021-01-16 18:19:34.079138+0200 0xc9       Default     0x0                  0      kernel: (AppleUSBEHCI) 005499.340799 EHC2@fa000000: AppleUSBEHCI::removePeriodicListEntry: unable to find isochronous list element <private> for frame 531
it restores playback after overload

Wouter1, 2021-01-16:
@roddy20  What do you mean "Official EMU driver". The one from Creative?

roddy20, 2021-01-16:
> @roddy20 What do you mean "Official EMU driver". The one from Creative?

уеs
a very old one (maybe for Lion)
it still works as a DAC for playback but no input recording

Wouter1, 2021-01-16:
@Smuteee  thanks for your nice words and suggestions. 
I like this device so I try to keep it running with as little work as possible. 
If you like you can also post on #109  :-)

Wouter1, 2021-01-16:
@roddy20 
> 16:06:53.845404 +0200 kernel 001458.004582 EHC2@fa000000: AppleUSBEHCI::removePeriodicListEntry: unable to find isochronous list element for frame 779

Yes this looks weird. I can't remember ever having seen these.  It seems a USB driver problem.

Tried several search engines, nobody seems to have reported this

is it possible that there is a hardware failure on the USB port or something like that?

Sometimes there are two separate USB chips on a computer, could you try a completely separate USB port or removing other USB devices fhat may be hogging the USB bus?

Smuteee, 2021-01-16:
@Wouter1 :  Will do ; )

roddy20, 2021-01-16:
> @roddy20
> 
 It seems a USB driver problem.

Yes
but port is the same
only driver differs
1) your driver writeCompleted bad result e00002ee and stop
2) their driver: a lot of AppleUSBEHCI::removePeriodicListEntry: unable to find isochronous list element for frame 531 but it begins to work again
I am not an advanced programmer, I don't know what is the difference inside driver
maybe they clear the buffer and start again on error?

> 
> Sometimes there are two separate USB chips on a computer, could you try a completely separate USB port or removing other USB devices fhat may be hogging the USB bus?

there is only EMU at that port
[image]
I ll try another port and let you know thanks

Wouter1, 2021-01-16:
@roddy20 if both orignal and new driver behave the same, it seems really a problem in the USB subsystem. Seems either a bug in USB driver or in the hardware

Here is a bug report about it
https://forum.blackmagicdesign.com/viewtopic.php?t=45261&p=262969

Apparently apple closed it but did not say whether they fixed it

I also reported a possibly related issue with the USB driver long ago (2014) in #18. Seems Apple never responded on my report.

Either way, Apple completely replaced their USB drivers with new drivers around that time. The whole subsystem  required me to completely rewrite the driver. This is all quite long ago, there is very little chance that anything can be fixed here.

Wouter1, 2021-01-16:
> I am not an advanced programmer, I don't know what is the difference inside driver. maybe they clear the buffer and start again on error?

I don't know, the original driver was very different, I had to rewrite significant parts to get it working decently.

Wouter1, 2021-01-16:
Fro the USB dump I see you have a 0202. There might be a very specific issue with 0202. I never had 0202 so I rely on user reports whether that worked.

roddy20, 2021-01-16:
> Either way, Apple completely replaced their USB drivers with new drivers around that time. The whole subsystem required me to completely rewrite the driver. This is all quite long ago, there is very little chance that anything can be fixed here.

it is possible
I cannot reproduce this error in Big Sur

roddy20, 2021-01-16:
I ll try your older v9 driver with Sierra, never did it before

Wouter1, 2021-01-16:
>it is possible I cannot reproduce this error in Big Sur

No, this error comes from a very old USB system. this USB system has been replaced very long ago.

Wouter1, 2021-01-16:
I think I got some version numbers messed up. Mavericks runs on darwin 13 but apparently is OSX 10.9

The v9 version is for OSX before osx 11 which I think refers to 10.11 El Capitan. Sierra is 10.12 (I thought it was earlier). So the v11 driver should be fit. The older one won't work then.

Wouter1, 2021-01-16:
So rethinking this, your sierra must be using the new USB system. I would expect it to properly support your USB ports. 

If you say it works fine on Big Sur, then that seems to rule out a hardware issue with the USB chips.

I have no idea or suggestions right here.

roddy20, 2021-01-16:
> If you say it works fine on Big Sur, then that seems to rule out a hardware issue with the USB chips.
> I have no idea or suggestions right here.

I think chips are Ok, other USB equipment works, Big Sur works, Maverick worked OK
EMU0202 works with its native driver, no recording, but playback is perfect

when I ll have a time I ll test Mojave or El Capitan
maybe it is the problem exactly in Sierra?

Wouter1, 2021-01-16:
@roddy20  

 "the original driver" is confusing me. FAIK the original driver never worked properly, neither recording nor playback. It crashed me continuously. I would think it can never work above 10.11 as the old USB system tat the original driver uses was not supported anymore. 

But if Big Sur works for you, why don't you just switch to Big Sur?

Possibly it is just a bug in the Sierra USB driver. But I would then at least expect your error to appear somewhere on the web. Maybe the error was there once,  but too long ago for Google to find it back?

roddy20, 2021-01-16:
> @roddy20
> 
> "the original driver" is confusing me. FAIK the original driver never worked properly, neither recording nor playback. It crashed me continuously. I would think it can never work above 10.11 as the old USB system tat the original driver uses was not supported anymore.

playback is good enough
recording is impossible at all, maybe Mavericks was the last (?)

roddy20, 2021-01-16:
> But if Big Sur works for you, why don't you just switch to Big Sur?

I am thinking about it
maybe later

there is a lot of incompatible software
I don't like new TV and Music, no iTunes
it cannot be installed to my Mac directly
my friend did clone of system drive for me

Wouter1, 2021-01-16:
"playback is good enough. recording is impossible at all, maybe Mavericks was the last (?)"

Sounds bizarre to me. To start with, playback can work only if recording works, or at least the datarate of the incoming recorded stream is tracked exactly which should basically amount to a working recorder.