Wouter1/EMU-driver #134: IRQ Problem on non-Apple macs.

jefazo2k5, 2021-01-18:
This issue affect all Macos versions.

The serial port has to be disabled in bios system, if enabled, the Os system detects wrong IRQ directions, that result on malfunction on system when you power on your audio interface.

Wouter1, 2021-01-18:
@jefazo2k5 what are you talking about, there  is no serial port nor a BIOS system on mac

jefazo2k5, 2021-01-18:
@Wouter1 this problem is  suffering If you use macos in non-apple  hardware.

Wouter1, 2021-01-18:
What are you trying to achieve with this ticket? I can not do anything about problems that hackintosh users may run into?

jefazo2k5, 2021-01-18:
)Sorry don't know what is correct place, i'm new in those enviroments.
i'm not talking about hackintosh, you can use macos on a virtual machine) is only info for other user that use your driver. Sorry for the incorvenence (and my english)

Wouter1, 2021-01-18:
@jefazo2k5  Yes, thanks for reporting. The info you give seems just a bit out of scope so I'm a bit puzzled. 

Do you want a hackintosh section somewhere?

Is it just that you wanted to report this and can I close this ticket now?

jefazo2k5, 2021-01-18:
@Wouter1  Is only info for non-apple users .You can close the ticket , or if you prefer , i can delete it.

Wouter1, 2021-01-18:
I created a section 'hackintosh notes' on the installation page. Thanks!

closed 2021-01-18 by Wouter1