Wouter1/EMU-driver #135: BIG SUR - 2errors

Antoha125, 2021-01-25:
Hello! I did everything according to the instructions but nothing works. Everything worked for Catalina before! Ask for help !!

E-MU 0204 

[image]

[image]

Wouter1, 2021-01-25:
@Antoha125  Did you approve using System Preferences? Did you get the pop-up?

Antoha125, 2021-01-26:
csrutil disable in recovery mode approved.
I did not receive a pop-up window, it is possible that after the permission was renewed, the map was working before the update

Wouter1, 2021-01-26:
@Antoha125  If you did not get the popup, you have to go to the security panel yourself to allow the extension to run 

Can you try that? System Preferences/Security&Privacy and click "Allow".

Antoha125, 2021-01-26:
repeated everything from the very beginning 
[image]
[image]

Wouter1, 2021-01-26:
@Antoha125  Why repeat?
Just do what I proposed

fotoknet, 2021-02-07:
in BIG SUR - SIP does not disabling that way.
after running installer script i just copied file /EMU-driver-master/v11/EMUUSBAudio.kext to folder /Volumes/EFI/EFI/CLOVER/kexts/Other
Now 0204 is working!!!

Wouter1, 2021-02-08:
@fotoknet  
So how do you disable sip on hackintosh?

fotoknet, 2021-02-08:
I didn’t turn it off. 
if i try to csrutil disable in recovery mode - after reboot csrutil status showed - all enable. 
i just put kext file in efi folder on efi partition

Wouter1, 2021-02-08:
@fotoknet  Yes I see. But I really would like to know how people can use the standard installer. And that requires sip disabled

closed 2021-03-26 by Wouter1