Wouter1/EMU-driver #136: ARM M1 issue Incompatible architecture: Cannot find arm64e in fat binary

bezromval, 2021-02-22:
I think its an end of en era. Finally I need to buy a new sound interface. The installer report is:
Error Domain=KMErrorDomain Code=3 "Error occurred unloading extensions: Missing extension with identifier com.emu.driver.EMUUSBAudio : Incompatible architecture: Cannot find arm64e in fat binary
Incompatible architecture: Binary is for x86_64, but needed arch arm64e" UserInfo={NSLocalizedDescription=Error occurred unloading extensions: Missing extension with identifier com.emu.driver.EMUUSBAudio : Incompatible architecture: Cannot find arm64e in fat binary
Incompatible architecture: Binary is for x86_64, but needed arch arm64e}
Error Domain=KMErrorDomain Code=71 "Incompatible architecture: Binary is for x86_64, but needed arch arm64e" UserInfo={NSLocalizedDescription=Incompatible architecture: Binary is for x86_64, but needed arch arm64e}
Waiting for the system to finish installation. Can take several minutes.
Executing: /usr/bin/kmutil install --volume-root / --check-rebuild
Rebuilding the kernel
Driver is installed and ready for use!
You can close this window.
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

Wouter1, 2021-02-22:
@bezromval  thanks for reporting. Yes I was expecting it.

I already tried to recompile the code but it does not seem to generate the ARM binary and spits errors where the code wes compiling in the previous XCode  And I don't have access to any ARM machine so even if it would compile I would be unable to test this

Wouter1, 2021-02-22:
------- notes for myself

Main error message I get from the compiler: 

```
TARGET_OS_OSX not defined
cdefs.h
In file included from ...EMUUSUAudio_info.c
```

Actually I'm i porting xmach/mach_types.h there. 
Strange, error comes not from my code but from a standard library.

Wouter1, 2021-02-22:
XCode says the file that causes the error is EMUUSBAudio_info.c
Ny project does not have such a file?
It does have a file EMUUSBAudio-info.plist. Is this something odd being generated wrong?

Wouter1, 2021-02-22:
Similar error to here
https://github.com/catchorg/Catch2/issues/2139
But no solution reported there

Wouter1, 2021-02-22:
https://stackoverflow.com/questions/12132933/preprocessor-macro-for-os-x-targets

gives more details but it's quite confusing

Wouter1, 2021-02-22:
INFOPLIST_PREPROCESS=YES is in the project settings (project.pbxproj). 
I think this creates the above mentioned file that includes cdef.h

If I uncheck that, the error remains because cdefs.h is still imported through

EMU-driver/USH.h -> IOKit/usb/StandardUSB.h -> IOKit->IOTypes.h -> IOKit/system.h -> sys/cdefs.h

Wouter1, 2021-02-22:
If I put 

#include <TargetConditionals.h>

right before the #include<IOKit/usb/StandardUSB.h>

then I get

IOKit/usb/IOUSBLog.h file not found. 

Maybe this is a step in the right direction

Wouter1, 2021-02-22:
After just removing that import, I'm back to the TERGET_OS_OSX not defined

But now through the import chain

EMUUSBAudio_info.c -> mach/mach_types.h->stdint.h -> machine/types.h -> i396/types.h -> sys/cdefs.h

Wouter1, 2021-02-22:
Again that is NOT my file. 
I have turned off the preprocess info.plist file. 
Also the "Create Info.plist section in binary" is off.

Wouter1, 2021-02-22:
This Derived Data seems wrong. 
Maybe I can find more about derivedData

https://vojtastavik.com/2018/09/02/what-is-inside-derived-data-xcode/ 

might help

Wouter1, 2021-03-14:
Is this an XCode configuration issue?

https://developer.apple.com/library/archive/documentation/DeveloperTools/Conceptual/cross_development/Configuring/configuring.html

Wouter1, 2021-03-14:
Well that first line there is already confusing. " you install Xcode, the installer creates a /Developer/SDKs directory"

I HAVE Xcode installed. But I don't have /Developer.

Wouter1, 2021-03-14:
The error disappears if I set in project the Apple Clang-Preprocessing / Preprocessor macros add TARGET_OS_OSX=1

But then a new error appears in EMUUSBAudioDevice.h line 249 {{{Invalid operands to binary expression ('nullptr_t' and 'UInt8')}}}

Wouter1, 2021-03-14:
If I change that line the NULL to 0 I get another error

```IOKit/usb/IOUSBLog.h' file not found```

Wouter1, 2021-03-14:
https://developer.apple.com/forums/thread/72248 
These files do not exist apparently above 10.10

Wouter1, 2021-03-14:
If I comment out that line, I get another error, 
EMUUSBAudioDevice.cpp ine 335

because audioEngine.init(NULL) apparently returns boolean. I was comparing it with NULL==.
Changed to !audioEngine.init(NULL).

After that, build succeeds! But not clear yet what that means, do we now have fat binary and how to test it

Wouter1, 2021-03-14:
In that building project, I now have "macOS Deployment Target" set to MacOS 10.12" and Base SDK macOS. 
But it also works if I set the deployment target to macOS 11.0.

Project format is XCode 3.2-compatible.

bezromval, 2021-03-14:
I didn't understand any word you write up here, but I'm waiting here for some beta driver for M1 and would like to test it on my MacBook Air with m1 chip and I already have proper usb type c printer cable for it. Just let me know here.

Wouter1, 2021-03-14:
@bezromval  thanks for offering help  on testing. I have no ARM machine so I can not test anything.

I have at this point something compiling on xcode 12.5 beta. FAIK this should support ARM compilation. 

But  It's not clear what the compiled driver is at this point. But given that it's 458kB size, strongly suggests it's not fat but just 1 architecture. That probably is intel as I could nowhere select any ARM compilation target...  

The notes above  are tech notes for myself, documenting what I do. I have not much time so occasionally I have a few hours. It's then essential that I can continue on, rather than redo, what I did last time. Hence all these notes.

Wouter1, 2021-03-14:
Maybe here
https://www.lewuathe.com/how-to-make-arm-cross-compilation-on-macos.html

Wouter1, 2021-03-14:
Yes, ```lipo -info .....kext/Contents/MacOS/EMUUSBAudio``` gives only x86_64 so not good yet.

Wouter1, 2021-03-14:
The required flags are in the project, under build settings/architecture.

The Architectures are set to Standard Arcitectures (Apple Silicon, Intel) so that looks good.

If I set "Build Active Architectures Only" to "Yes" it builds fine.
I had that setting for the Debug mode; and aparently Build builds the debug version...

But if I set it to No then I get another error

EMUUSBAudioEngine line 47
```Allocating an object of abstract class type 'EMUUSBAudioEngine``` (?? looks like a conclusion, not an error message)

Wouter1, 2021-03-14:
Maybe the class EMUUSBAudioEngine does not implement one of the many virtual functios of the superclass?

I see differences, for instance 
performFormatChange/4 while it was /3 
getStatusDescriptor/1 I don't see it in imple
getNearestStartTime I don't see it in imple

How does this work, why does it compile for debug but not for run?

Wouter1, 2021-03-14:
Maybe it compiles ok on intel but not on arm? If you switch the "build active architecture only' to 'yes' for debug mode, and if build by default only copmpiles the debug version?

Wouter1, 2021-03-14:
Even weirder, the error ALSO shows in intel. However in spite of the error, the build succeeds there anyway.

Wouter1, 2021-03-14:
There is more detail on the error message in the issue navigator on the left.
Unfortunately it's very hard to find that detail as there is a huge list of warnings, it's completely hidden between those.

it says

```unimplemented virtual method driverDesiredHiResSampleIntervals in EMUUSBAudioEngine```

Wouter1, 2021-03-14:
The doc actually says "this is a new method which only exists on ARM platforms and therefore breaks binary compatibility if this is compiled on intel platforms"..... Well at least they have documented this.

Wouter1, 2021-03-14:
Added the missing function, only for ARM. It now compiles.
Created new M1 branch and pushed the new code there, including the new kext version v15

Wouter1, 2021-03-14:
@bezromval 
I just created a first version of the M1 driver. I have no idea what it will do, so please be careful. Maybe run from a separate partition just in case.

There is a new M1 branch containing a new version of the driver : v15

https://github.com/Wouter1/EMU-driver/tree/M1

It's in the v15 directory.

The installation script is NOT recognising that new file yet.

Can you install the v15 driver manually in /Library/Extensions and see what it does?

bezromval, 2021-03-15:
Thanks for spending your free time fo this! Ive just tried to install driver. What I've done:
1 boot in recovery mode, lunch terminal and use command csrutil disable. It disabled successful.Reboot Mac.
2 copy new kext v15 to the /Library/Extensions
3 In the system preferences/security and privacy allow this action and system cache was rebuild automatically.
4 Reboot system
5 Sound card did not appear in system preferences/sound

Wouter1, 2021-03-15:
@bezromval  can you please report any error messages and relevant console reports?
I need to know what is going wrong

bezromval, 2021-03-15:
Since I was doing it manually there is no error reports. What exactly I should to do to get report in the end?

bezromval, 2021-03-15:
Now I've tried: sudo kextcache -I / in terminal and here is the message

Executing: /usr/bin/kmutil install --volume-root / --check-rebuild
Not rebuilding KC for arch: 'arm64e'
kmutil done

Wouter1, 2021-03-15:
Just checking, you got a prompt after copying the kext into /Library/Extensions? What did it say?

thanks for checking the message. 

" Not rebuilding KC for arch: 'arm64e'" mmm not very informative.

regarding the log

https://github.com/Wouter1/EMU-driver/blob/master/FAQ.md and check "How do I make a system log of a mal-functioning driver?".

Wouter1, 2021-03-15:
Trying what it does on the test laptop.

I also need to do 
```
sudo chown root EMUUSBAudio.kext
sudo chgrp -R wheel EMUUSBAudio.kext
```

But then I still get "Invalid ownership (501:0) should be (0:0)

Wouter1, 2021-03-15:
And itl works on my test laptop!

Wouter1, 2021-03-15:
@bezromval  maybe something went wrong when you copied it to /Library/Extensions.

Did you use ```sudo cp -R EMUUSBAudio.kext /Library/Extensions/EMUUSBAudio.kext```?

Did you get the prompt "System extension updated. You must approve it"?

Did you approve the kext ?

Did you reboot after approving?

bezromval, 2021-03-16:
I’ve tried everything. It did work for me. I don’t know why. I even tried kext-Droplet.app.

bezromval, 2021-03-16:
Using sudo cp -R EMUUSBAudio.kext /Library/Extensions/EMUUSBAudio.kext
EMUUSBAudio.kext is in extensions folder after that. There is NO any prompt System extension updated or something. But Iam rebooting anywhere. There is no sound card appears in settings.
Prompt appears only if I delete kext from folder. Ask me reboot and is rebuilding cache.

Wouter1, 2021-03-16:
@bezromval  it's not just copying. All permissions have to be right. Only if it's all right you get the prompt. Rebooting is no use until you get that prompt. It means it's not accepted if you don't get the prompt.

It's strange that a prompt appears if you *remove*. But I did not try. 

Of course v15. This is the new folder. I did not change the old folders.

What does ```ls -al /Library/Extensions``` show after the sudo cp?

What does  ```sudo kextload /Library/Extensions/EMUUSBAudio.kext``` show?

Wouter1, 2021-03-16:
@bezromval  you can also try 
```sudo kextutil -t  /Library/Extensions/EMUUSBAudio.kext```
what does it show?

bezromval, 2021-03-17:
Last login: Wed Mar 17 17:36:00 on console
bezromval@MacBook-Air v15 % cp -R EMUUSBAudio.kext /Library/Extensions/EMUUSBAudio.kext
bezromval@MacBook-Air v15 % 
  [Восстановлен 17 марта 2021 г., 17:39:12]
Last login: Wed Mar 17 17:39:10 on console
Restored session: среда, 17 марта 2021 г. 17:37:41 (MSK)
bezromval@MacBook-Air v15 % ls -al /Library/Extensions
total 0
drwxr-xr-x   9 root       wheel   288 16 мар 21:44 .
drwxr-xr-x  65 root       wheel  2080 10 мар 19:49 ..
drwxrwxr-x@  4 bezromval  wheel   128 17 мар 17:36 EMUUSBAudio.kext
drwxr-xr-x   3 root       wheel    96  1 янв  2020 HighPointIOP.kext
drwxr-xr-x   3 root       wheel    96  1 янв  2020 HighPointRR.kext
drwxr-xr-x   3 root       wheel    96 23 янв 02:50 SATSMARTDriver.kext
drwxr-xr-x   3 root       wheel    96 11 сен  2019 SATSMARTLib.plugin
drwxr-xr-x   3 root       wheel    96  6 дек  2017 SONYDeviceType04.kext
drwxr-xr-x   3 root       wheel    96  1 янв  2020 SoftRAID.kext
bezromval@MacBook-Air v15 % kextload /Library/Extensions/EMUUSBAudio.kext
Executing: /usr/bin/kmutil load -p /Library/Extensions/EMUUSBAudio.kext
Error Domain=KMErrorDomain Code=71 "Insufficient permissions for action: Unable to load 'com.binaryfruit.driver.SATSMARTDriver' in unprivileged request." UserInfo={NSLocalizedDescription=Insufficient permissions for action: Unable to load 'com.binaryfruit.driver.SATSMARTDriver' in unprivileged request.}
bezromval@MacBook-Air v15 % sudo kextutil -t /Library/Extensions/EMUUSBAudio.kext
Password:
Executing: /usr/bin/kmutil load --bundle-path /Library/Extensions/EMUUSBAudio.kext
Error Domain=KMErrorDomain Code=30 "Validating extension failed: Kext com.emu.driver.EMUUSBAudio v4.1.0 in executable kext bundle com.emu.driver.EMUUSBAudio at /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext:

Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/Contents
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/Contents/MacOS
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/Contents/MacOS/EMUUSBAudio
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/Contents/Resources
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/Contents/Resources/en.lproj
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/Contents/Resources/en.lproj/InfoPlist.strings
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/Contents/Info.plist
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext/Contents
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext/Contents/MacOS
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext/Contents/MacOS/EMUUSBAudio
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext/Contents/Resources
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext/Contents/Resources/en.lproj
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext/Contents/Resources/en.lproj/InfoPlist.strings
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext/Contents/Info.plist
Authenticating extension failed: Bad code signature" UserInfo={NSLocalizedDescription=Validating extension failed: Kext com.emu.driver.EMUUSBAudio v4.1.0 in executable kext bundle com.emu.driver.EMUUSBAudio at /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext:

Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext
[... cut ...]
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext/Contents/Resources/en.lproj/InfoPlist.strings
Filesystem error: Invalid ownership (501:0) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.emu.driver.EMUUSBAudio.OW7oqr/EMUUSBAudio.kext/EMUUSBAudio.kext/Contents/Info.plist
Authenticating extension failed: Bad code signature}
bezromval@MacBook-Air v15 %

bezromval, 2021-03-17:
Last login: Wed Mar 17 17:50:26 on console
bezromval@MacBook-Air v15 % csrutil status
System Integrity Protection status: disabled.
bezromval@MacBook-Air v15 % sudo cp -R EMUUSBAudio.kext /Library/Extensions/EMUUSBAudio.kext
Password:
bezromval@MacBook-Air v15 % ls -al /Library/Extensions
total 0
drwxr-xr-x   9 root  wheel   288 17 мар 17:51 .
drwxr-xr-x  65 root  wheel  2080 10 мар 19:49 ..
drwxr-xr-x@  3 root  wheel    96 17 мар 17:51 EMUUSBAudio.kext
drwxr-xr-x   3 root  wheel    96  1 янв  2020 HighPointIOP.kext
drwxr-xr-x   3 root  wheel    96  1 янв  2020 HighPointRR.kext
drwxr-xr-x   3 root  wheel    96 23 янв 02:50 SATSMARTDriver.kext
drwxr-xr-x   3 root  wheel    96 11 сен  2019 SATSMARTLib.plugin
drwxr-xr-x   3 root  wheel    96  6 дек  2017 SONYDeviceType04.kext
drwxr-xr-x   3 root  wheel    96  1 янв  2020 SoftRAID.kext
bezromval@MacBook-Air v15 % sudo kextload /Library/Extensions/EMUUSBAudio.kext
Executing: /usr/bin/kmutil load -p /Library/Extensions/EMUUSBAudio.kext
Error Domain=KMErrorDomain Code=31 "Error occurred while building a collection: 
    1: One or more binaries has an error which prevented linking.  See other errors.
    2: Could not use 'com.emu.driver.EMUUSBAudio' because: Failed to bind '__ZN8IOSyncer6createEb' in 'com.emu.driver.EMUUSBAudio' (at offset 0x1e0 in __DATA_CONST, __got) as could not find a kext which exports this symbol
com.emu.driver.EMUUSBAudio specific: 
    1: Failed to bind '__ZN8IOSyncer6createEb' in 'com.emu.driver.EMUUSBAudio' (at offset 0x1e0 in __DATA_CONST, __got) as could not find a kext which exports this symbol
" UserInfo={NSLocalizedDescription=Error occurred while building a collection: 
    1: One or more binaries has an error which prevented linking.  See other errors.
    2: Could not use 'com.emu.driver.EMUUSBAudio' because: Failed to bind '__ZN8IOSyncer6createEb' in 'com.emu.driver.EMUUSBAudio' (at offset 0x1e0 in __DATA_CONST, __got) as could not find a kext which exports this symbol
com.emu.driver.EMUUSBAudio specific: 
    1: Failed to bind '__ZN8IOSyncer6createEb' in 'com.emu.driver.EMUUSBAudio' (at offset 0x1e0 in __DATA_CONST, __got) as could not find a kext which exports this symbol
}
bezromval@MacBook-Air v15 % sudo kextutil -t /Library/Extensions/EMUUSBAudio.kext
Executing: /usr/bin/kmutil load --bundle-path /Library/Extensions/EMUUSBAudio.kext
Error Domain=KMErrorDomain Code=31 "Error occurred while building a collection: 
    1: One or more binaries has an error which prevented linking.  See other errors.
    2: Could not use 'com.emu.driver.EMUUSBAudio' because: Failed to bind '__ZN8IOSyncer6createEb' in 'com.emu.driver.EMUUSBAudio' (at offset 0x1e0 in __DATA_CONST, __got) as could not find a kext which exports this symbol
com.emu.driver.EMUUSBAudio specific: 
    1: Failed to bind '__ZN8IOSyncer6createEb' in 'com.emu.driver.EMUUSBAudio' (at offset 0x1e0 in __DATA_CONST, __got) as could not find a kext which exports this symbol
" UserInfo={NSLocalizedDescription=Error occurred while building a collection: 
    1: One or more binaries has an error which prevented linking.  See other errors.
    2: Could not use 'com.emu.driver.EMUUSBAudio' because: Failed to bind '__ZN8IOSyncer6createEb' in 'com.emu.driver.EMUUSBAudio' (at offset 0x1e0 in __DATA_CONST, __got) as could not find a kext which exports this symbol
com.emu.driver.EMUUSBAudio specific: 
    1: Failed to bind '__ZN8IOSyncer6createEb' in 'com.emu.driver.EMUUSBAudio' (at offset 0x1e0 in __DATA_CONST, __got) as could not find a kext which exports this symbol
}
bezromval@MacBook-Air v15 %

bezromval, 2021-03-17:
Hope, you know what all this stuff means because I'm totally not.

Wouter1, 2021-03-17:
@bezromval thanks!
It looks like some libraries, or specific versions of libraries, do not exist anymore

Strange, they do exist apparently on the intel version of Big Sur, but not on the M1 version! Just to be sure, you are also on Big Sur?

bezromval, 2021-03-17:
Iam on Big Sur 11.2.3. Its the only version can be installed on m1.

philippgadow, 2021-03-17:
Hi! Thank you for providing the driver which allowed me to use the EMU tracker pre interface until today. I am very happy that it exist and you keep updating it.

On my work laptop, which is a M1 MBP with BigSur 11.1 (20C69), I can confirm that I get exactly the same errors as bezromval.

Wouter1, 2021-03-18:
The main use is inside EMUUSBAudioEngine start() 
An IOSyncer called mSyncer is created.

The function of this is not clear to me.
IOSyncer seems to have been deprecated already long ago. I can't find any documentation about it.
It seems to be used to sync the registration of some IOServiceNotificationHandler. Which is then immediately removed as well.
There is a call to mSyncer.signal in the audioDevicePublished method, after found the audio engine to be used. mSyncer.wait could be where it waits for signal.

Maybe this was a mechanism to get the start() to wait for the audioDevicePublished event?

Wouter1, 2021-03-19:
https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/commit/b33b10c0a6df2e4474d3acc057a6a4f602bf79a1

" //TODO: IOSyncer is deprecated in 10.7 headers, gone from 10.8 headers"

Wouter1, 2021-03-19:
But IOSyncer,cpp is in my source tree. Why is the symbol not resolved?

Wouter1, 2021-03-19:
The cpp code was not part of the project. Apparently I once removed it because it was not needed (but hept the sources where they were)

Wouter1, 2021-03-19:
Tried to add them. But now the driver does not load

I first get OSMetaClass: Kext com.emu.driver.EMUUSBAudio class IOSyncer is a duplicate; kext __ kernel__ already has a class by that name.
kext ... did not start (return code 0xdc00400a
failed to load 0xdc008017
Couldn't alloc class "EMUUSBAudioDevice"

Wouter1, 2021-03-19:
Trying to rename the whole copy to IOSyncer2 to avoid name collision

Wouter1, 2021-03-19:
It works on my test machine. Let's give it a try on M1

Wouter1, 2021-03-19:
@bezromval I updated the code, hopefully that adresses the issue that appeared in the output that you posted. Can you retry?

bezromval, 2021-03-19:
e-mu appears in sound devices but output not working and there is no any controls in output. And when i turned off e-mu there was some panic with this report after rebooting:

panic(cpu 3 caller 0xfffffe001b52ff94): "lck_mtx_assert(): mutex (0xfffffe2335631e90) not owned"
Debugger message: panic
Memory ID: 0x6
OS release type: User
OS version: 20D91
Kernel version: Darwin Kernel Version 20.3.0: Thu Jan 21 00:06:51 PST 2021; root:xnu-7195.81.3~1/RELEASE_ARM64_T8101
Fileset Kernelcache UUID: F78A48EC84D7C3283E7C6C67D00889A7
Kernel UUID: 9FE8C0DA-8ED0-381C-9CEC-2A779F3E1503
iBoot version: iBoot-6723.81.1
secure boot?: YES
Paniclog version: 13
KernelCache slide: 0x00000000137d0000
KernelCache base:  0xfffffe001a7d4000
Kernel slide:      0x0000000014304000
Kernel text base:  0xfffffe001b308000
Kernel text exec base:  0xfffffe001b3d0000
mach_absolute_time: 0x7f69c3a5
Epoch Time:        sec       usec
  Boot    : 0x6054b5c2 0x0004b1bf
  Sleep   : 0x00000000 0x00000000
  Wake    : 0x00000000 0x00000000
  Calendar: 0x6054b614 0x000b025d

CORE 0 recently retired instr at 0xfffffe001b53ac0c
CORE 1 recently retired instr at 0xfffffe001b53ac0c
CORE 2 recently retired instr at 0xfffffe001b53ac0c
CORE 3 recently retired instr at 0xfffffe001b539748
CORE 4 recently retired instr at 0xfffffe001b53ac10
CORE 5 recently retired instr at 0xfffffe001b53ac10
CORE 6 recently retired instr at 0xfffffe001b53ac10
CORE 7 recently retired instr at 0xfffffe001b53ac10
Panicked task 0xfffffe166e106550: 2021 pages, 11 threads: pid 437: coreaudiod
Panicked thread: 0xfffffe16730b0cb0, backtrace: 0xfffffe3fe1143490, tid: 11379
		  lr: 0xfffffe001b41dfd0  fp: 0xfffffe3fe1143500
		  lr: 0xfffffe001b41dd9c  fp: 0xfffffe3fe1143570
		  lr: 0xfffffe001b53ff0c  fp: 0xfffffe3fe1143590
		  lr: 0xfffffe001b5318b8  fp: 0xfffffe3fe1143640
		  lr: 0xfffffe001b3d77e8  fp: 0xfffffe3fe1143650
		  lr: 0xfffffe001b41da2c  fp: 0xfffffe3fe11439e0
		  lr: 0xfffffe001b41da2c  fp: 0xfffffe3fe1143a50
		  lr: 0xfffffe001bbb865c  fp: 0xfffffe3fe1143a70
		  lr: 0xfffffe001b52ff94  fp: 0xfffffe3fe1143a90
		  lr: 0xfffffe001ba8d69c  fp: 0xfffffe3fe1143ab0
		  lr: 0xfffffe001a6c09e0  fp: 0xfffffe3fe1143af0
		  lr: 0xfffffe001b4fe7d8  fp: 0xfffffe3fe1143b50
		  lr: 0xfffffe001b422fb0  fp: 0xfffffe3fe1143bc0
		  lr: 0xfffffe001b3fb9dc  fp: 0xfffffe3fe1143c40
		  lr: 0xfffffe001b424124  fp: 0xfffffe3fe1143c70
		  lr: 0xfffffe001b40aff0  fp: 0xfffffe3fe1143d10
		  lr: 0xfffffe001b414bf4  fp: 0xfffffe3fe1143d50
		  lr: 0xfffffe001b4126e8  fp: 0xfffffe3fe1143d70
		  lr: 0xfffffe001b5252d8  fp: 0xfffffe3fe1143e40
		  lr: 0xfffffe001b53192c  fp: 0xfffffe3fe1143ef0
		  lr: 0xfffffe001b3d77e8  fp: 0xfffffe3fe1143f00
      Kernel Extensions in backtrace:
         com.emu.driver.EMUUSBAudio(4.1)[793BBAA1-B3AA-3CED-A418-9FB2AA150B5D]@0xfffffe001a6a4000->0xfffffe001a6cffff
            dependency: com.apple.iokit.IOAudioFamily(300.6.1)[8F3BCABA-5A95-3E59-BCA4-563D59AC1339]@0xfffffe001d2dc000->0xfffffe001d303fff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[60A0F164-C469-34AD-A787-F64C60948655]@0xfffffe001db94000->0xfffffe001dc4bfff

last started kext at 746195861: com.apple.filesystems.autofs	3.0 (addr 0xfffffe001b280000, size 16384)
loaded kexts:
com.emu.driver.EMUUSBAudio	4.1.0
com.apple.filesystems.autofs	3.0
com.apple.fileutil	20.036.15
com.apple.driver.AppleBluetoothMultitouch	99
com.apple.driver.AppleTopCaseHIDEventDriver	4030.5
com.apple.iokit.IOBluetoothSerialManager	8.0.3d9
com.apple.driver.AppleBiometricServices	1
[... 193 lines cut ...]
com.apple.iokit.IOAVFamily	1.0.0
com.apple.iokit.IOHDCPFamily	1.0.0
com.apple.iokit.IOCECFamily	1
com.apple.iokit.IOAudio2Family	1.0
com.apple.driver.AppleEmbeddedAudioLibs	1.17
com.apple.driver.AppleFirmwareUpdateKext	1
com.apple.driver.AppleM2ScalerCSCDriver	265.0.0
com.apple.iokit.IOSurface	289.3
com.apple.driver.IODARTFamily	1
com.apple.security.quarantine	4
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.kext.CoreTrust	1
com.apple.security.AppleImage4	3.0.0
com.apple.iokit.IOCryptoAcceleratorFamily	1.0.1
com.apple.driver.AppleARMPlatform	1.0.2
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.IOReportFamily	47
com.apple.kec.pthread	1
com.apple.kec.corecrypto	11.1
com.apple.kec.Libm	1

** Stackshot Succeeded ** Bytes Traced 268266 (Uncompressed 678560) **

Wouter1, 2021-03-19:
@bezromval  thanks for the report.
So the driver now loads. We're heading in the right direction.

If I understand correctly, your machine only crashed when you turned off the emu.  That gives more possibilities to see what is going on.

Can you report the console output and post the EMUUSB messages? Don't turn off the driver until you have copied the text into a file.

Wouter1, 2021-03-19:
If the IOSyncer remains an issue, we may need other syncing/locking. https://flylib.com/books/en/3.126.1.120/1/

Wouter1, 2021-03-19:
@philippgadow  thanks for confirming.

bezromval, 2021-03-20:
Last login: Sat Mar 20 15:20:20 on console
bezromval@MacBook-Air ~ % sudo kextload /Library/Extensions/EMUUSBAudio.kext
Password:
Executing: /usr/bin/kmutil load -p /Library/Extensions/EMUUSBAudio.kext
bezromval@MacBook-Air ~ % sudo kextutil -t /Library/Extensions/EMUUSBAudio.kext
Executing: /usr/bin/kmutil load --bundle-path /Library/Extensions/EMUUSBAudio.kext
bezromval@MacBook-Air ~ %

[34 more replies]