Wouter1/EMU-driver #137: system integrity re-enable driver stops working

thelostprophet21, 2021-03-29:
I don't like the ideea of leaving the system integrity disabled. It works great with it disabled but after I re-enable, it does not detect the soundcard. Any thoughts on this?

Wouter1, 2021-03-29:
@thelostprophet21 

Yes. My contribution to the world is software for EMU users.

I don't agree with Apple SIPping my money through periodic fees for this. It's like a penalty on work done for free.

I'm also not convinced that this SIP policy helps "system integrity", as you basically need money, not integrity, to get such a signing key. And maybe you don't even need a key to get malicious software into the mac kernel.

More in detail: 
Apple's sells software signing keys, and only by using that, OSX SIP accepts my driver. 
Thus they force developers buying thesekeys

For now they still offer the possibility to turn off this SIP mechanism. This may change in the future and then using my driver would be made impossible by Apple.

You have any suggestions to improve the situation?

thelostprophet21, 2021-03-29:
Thanks for your reply. I am not a programmer and am fairly new to MacOS and
felt like I am doing something wrong if I leave that security option turned
off.

Wouter1, 2021-03-30:
@thelostprophet21 nothing wrong with that. 
But it makes it easier for other software too to put something into your kernel. So just be a bit  careful on what you download and run

closed 2021-03-30 by Wouter1

thelostprophet21, 2021-03-30:
Thanks a lot for your help and support! Have a great year ahead.

thelostprophet21, 2021-04-26:
Hello. I've just updated to Big Sur OS and was surprised to see an update
on EMU github as well. Thanks a lot for your support. I can still use the
emu because of you.
Is there any way I could send a donation? I would like to buy a cup of
coffee :)
Thanks

Andrei

Wouter1, 2021-04-26:
@thelostprophet21 

Thanks for the report! Glad that you like it.

You could report the good working and thanks on the ticket for that -  https://github.com/Wouter1/EMU-driver/issues/109

I don't know you could give me a money  donation. I  never bothered with internet payment systems and know very little about that Sending money with snail mail would probably cost more than your intended donation to start with...

thelostprophet21, 2021-04-27:
Ever thought of posting a BTC, or ETH address? It's almost free, takes 5
min to setup, and cryptocurrency is increasing in popularity.

Wouter1, 2021-04-27:
@thelostprophet21  I know what they are and how they work in theory  but never looked on how they really work in practice.