Wouter1/EMU-driver #138: Working Properly - Big Sur

thelostprophet21, 2021-04-27:
Hello. For me it's working perfectly. I'm using the sample rate at 48khz with the EMU 0202.
Small note: I had to open the installer with terminal in order to have it run.

macOS BigSur v 11.2.3

2.4 Ghz 8- Core Intel i9

Wouter1, 2021-04-27:
@thelostprophet21  thanks for reporting! 
I posted your thanks in #109 which is the proper place for this

closed 2021-04-27 by Wouter1

fotoknet, 2021-05-03:
worked in 11.2 and has broken in 11.3 :(
listed in usb devices, but disappeared in audio devices after OS update

thelostprophet21, 2021-05-24:
I have Big Sur 11.2.3 and not updated because of your comment. Do you have an update?
What version have you updated to exactly?

Wouter1, 2021-05-24:
@totoknet this ticket is closed and it's only about thelostprophet reporting the driver works for him on Big Sur. 
My comment to him was that reports for proper working should be on #109