Wouter1/EMU-driver #140: OSX Monterey E-MU 0404 USB Success

iRet, 2023-01-19:
Everything functional. 
Sharing my experience as it might be useful for someone else.

### The process:
I had to remove some obsolete and unused kext hanging around. 
For me it was Nvidia web driver and HoRNDIS. 
I think it has to do with the installer script.
After this the installation succeeded. 

I prefer to disable SIP only for kext.
1. In recovery terminal: ```csrutil disable```
2. Reboot and install driver with shell script
3. In recovery terminal: ```csrutil enable --without kext```
4. Reboot and check: ```csrutil status```
> 	Apple Internal: disabled
> 	Kext Signing: disabled
> 	Filesystem Protections: enabled
> 	Debugging Restrictions: enabled
> 	DTrace Restrictions: enabled
> 	NVRAM Protections: enabled
> 	BaseSystem Verification: enabled

### Additional steps:
It was asking me to allow **EMUMIDIDriver.plugin** every time I start DAW.
In order to fix I ran: ```sudo xattr -rd com.apple.quarantine /Library/Audio/MIDI\ Drivers/EMUMIDIDriver.plugin```

jkwinta, 2024-04-28:
Do you know/remember what kext you did remove?
Or how you determined that?

jkwinta, 2024-04-28:
Nevermind, it looks like it runs from the executable inside the package, but not from the package itself, hmm....

bbatarelo, 2026-09-04:
Driver for Apple Silicon: https://github.com/bbatarelo/emu-apple-silicon